chipKIT32/chipKIT-importer #1: Importing for WiFire with NoCopy does not work

EmbeddedMan, 2017-02-14:
This issue is meant to keep all of the relevant pieces of information and conversation about this particular issue in one place, rather than a long email chain that's also being used for many other things.

Here are the various pieces of the problem, as copied from the emails:

======
Guys, I have a problem with GCC

It turns out that when importing a project for WiFire without copying files, the Board_Data.c file from the variant directory is not used by the compiler when resolving "includes". Instead, the compiler always uses the core version which does not work for WiFire. When building the libCore.a library:

"/opt/microchip/xc32/v1.43-TC5/bin/xc32-g++" -c  -g -x c -w -O1 -Wall -Wextra -DARDUINO_ARCH_PIC32 -mno-smart-io -ffunction-sections -fdata-sections -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD  -mnewlib-libc -mprocessor=32MZ2048EFG100 -I"/home/gregor/.arduino15/packages/chipKIT/hardware/pic32/1.3.1/variants/WiFire" -I"/home/gregor/.arduino15/packages/chipKIT/hardware/pic32/1.3.1/cores/pic32" -DF_CPU=200000000UL -DARDUINO=10613 -D_BOARD_WIFIRE_ -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino -DXPRJ_default=default -D__CTYPE_NEWLIB "/home/gregor/.arduino15/packages/chipKIT/hardware/pic32/1.3.1/cores/pic32/pins_arduino.c" -o "pins_arduino.c.o"

I'm getting:

In file included from /home/gregor/.arduino15/packages/chipKIT/hardware/pic32/1.3.1/cores/pic32/pins_arduino.c:85:0:
/home/gregor/.arduino15/packages/chipKIT/hardware/pic32/1.3.1/cores/pic32/Board_Data.c: In function 'altSDchipSelectHigh':
/home/gregor/.arduino15/packages/chipKIT/hardware/pic32/1.3.1/cores/pic32/Board_Data.c:638:22: error: '_EXTERNAL_1_IRQ' undeclared (first use in this function)
      restoreIntEnable(_EXTERNAL_1_IRQ, interrupt_state);
                       ^

It's clear from this error message that the compiler uses the "core" Board_Data.c and not the one in "variant" directory despite the fact that the "variant" directory is passed as -I option before the "core". Any ideas how to convince the compiler to use the "variant" version?

gh
=============

=============
Compiler expert Jason Kajita explains:
 
For the quote form of #include “file”, the preprocessor will look in the same directory as the .c file first. So in this case, the preprocessor will search the same directory as pins_arduino.c first.
 
You should be able to use the angle-bracket form of #include <file>. This will cause the preprocessor to look in the -I directories first, and then the standard system directories.
 
G
=============

=============
The chipKIT/hardware/pic32/1.3.1/cores/pic32/pins_arduino.c file contains:

#include "Board_Data.c"

which, according to what Jason wrote, eliminates the possibility of any customization by placing Baord_Data.c in the variant directory (which is the case in WiFire and few other boards). I can handle this in the import plugin by copying all core and variant files into a single, temporary directory and then building the libCore.a but it seems to me that this is a bug in chipKIT core.

gh
=============

EmbeddedMan, 2017-02-14:
I'm starting to investigate this a little bit, and I'm starting to get confused.

All chipKIT variants have Board_Data.c files in their variant folders, and all sketches depend on that file being compiled into the final build. So why does this problem only occur with WiFire? What's different about it that causes other boards to work in NoCopy mode, but WiFire to fail?

And why does this process work fine for Arduino IDE, but not for chipKIT Importer? What are we doing differently that's causing it to fail?

gholdys, 2017-02-14:
Actually, I've never managed to build any sketch for WiFire on Arduino so I can't compare.

EmbeddedMan, 2017-02-14:
Well, that's a problem! The stock Arduino IDE 1.8.1 + chipKIT-core v1.3.1 should have no problem building sketches for the WiFire series of boards. Let's get that working first, so that you can see exactly how Arduino Builder is able to get WiFire working. Can you uninstall and reinstall Arduino IDE and chipKIT-core to see if that fixes WiFire building for you?

gholdys, 2017-02-14:
Sure...

gholdys, 2017-02-14:
Yeah, that helped...

gholdys, 2017-02-14:
It seems that you guys experienced a completely different error than me. I'm sending a new version of the plugin via email...

gholdys, 2017-02-15:
I've created a "dist" directory in out repo and committed a newly built nbm file. Please try it out.

EmbeddedMan, 2017-02-15:
It appears to me that the latest importer (v0.9, Feb 14th) works properly
and fixes this WiFire issue. I now have no problem both importing and
building all of my test sketches (in NoCopy and Copy modes) for WiFire.

gholdys, 2017-02-16:
Should we close this issue?

EmbeddedMan, 2017-02-16:
Yes, it works for me quite well now. I'll open up another issue if I find
any other problems.

closed 2017-02-16 by gholdys