chipKIT32/chipKIT-importer #2: "Program device" button greys out and doesn't come back

EmbeddedMan, 2017-02-15:
With the current (v0.9, Feb 14th) version of the importer, if I import the blink sketch for Fubarino Mini, and then put the Fubarino Mini into program mode, select the proper COM port in the pull down, and then click the CK button, everything works great. The FB Mini is programmed just fine.

However, if I then import another sketch for FB Mini, built it, put the FB Mini into program mode, the CK button is greyed out, and I can't click on it to perform the download.

If I then click on a different project in the Projects list in MPLAB X, then back on the just-built project, the CK  button becomes active again and I can click on it. But sometimes that doesn't work either. It depends on if the FB Mini is in bootloader mode or not.

After more testing, I think that what's happening is that the background scanning of available COM ports is not updating the CK button properly. If there are no available COM ports (which happens after programming the FB Mini with a sketch that does not invoke the USB serial object) then the CK button goes grey and can't be clicked, which is correct. But then when I put the FB Mini into bootloader mode, the COM port exists again, but the importer code doesn't notice it properly and enable the CK button until I switch projects in the Projects list.

gholdys, 2017-02-15:
Thanks, I'll check this.

gholdys, 2017-04-26:
Is this issue still valid?

EmbeddedMan, 2017-04-26:
I believe it still is, but I need to confirm.

EmbeddedMan, 2017-04-26:
I have confirmed that this is still a problem with the latest (v1.0.1) importer. Even though my Mini is in bootloader mode, and the COM port exists, the plugin doesn't see it. I click on the port dropdown and there are no ports listed, and the CK button stays grayed out.

EmbeddedMan, 2017-04-26:
Ahh! I just found out a clue.

Normally my PC just has a hardware COM1 port available. When I have a FB Mini in bootloader mode, I gets a second port, COM22.

Now, when the above problem happens, I click the pulldown, and no COM ports are shown. BUT! There is _space_ for two com ports. It's like it knows there are two COM ports, but it's not actually printing them into the pulldown.

If I then click on the _empty_ spot in the pulldown where COM22 should be, then everything works great. The CK button lights up, and I can program again.

So the plugin knows both ports are there, but it's not drawing them into the pulldown.

EmbeddedMan, 2017-04-26:
Also, once I click in the right spot in the pulldown, and the CK button lights up, then I see "COM22" shown in the pulldown, and if I click it again, I see both COM1 and COM22 just like there should be. 

So this sounds like some sort of a refresh/redraw problem.

gholdys, 2017-04-27:
That's useful information. Thanks.