chipKIT32/chipKIT-importer #3: pic32prog invoked when Build Project button clicked

EmbeddedMan, 2017-02-15:
After playing around with importing the blink sketch for FB Mini and downloading it to the board, I came across the following behavior: When I click the Build Project button (just the hammer), MPLAB X tries to program my board. Here is the output of the Program Device tab in the Output window (Note that the FB Mini was not in program mode, as I did not want to program the board, just build the project.):

-- Running native process in default directory: "C:\Users\bschmalz\AppData\Local\Arduino15\packages\chipKIT\tools\pic32prog\v2.1.24\pic32prog.exe" -d "COM17" -b "115200" "C:/Users/bschmalz/MPLABXProjects/blink_FBMini_Copoy.X/dist/default/production/blink_FBMini_Copoy.X.production.hex" --
Programmer for Microchip PIC32 microcontrollers, Version 2.1.24
    Copyright: (C) 2011-2015 Serge Vakulenko
\\.\COM17: Cannot open

No target found.
-- Running native process in default directory: "C:\Users\bschmalz\AppData\Local\Arduino15\packages\chipKIT\tools\pic32prog\v2.1.24\pic32prog.exe" -d "COM17" -b "115200" "C:/Users/bschmalz/MPLABXProjects/blink_WiFire_NoCopy.X/dist/default/production/blink_WiFire_NoCopy.X.production.hex" --
Programmer for Microchip PIC32 microcontrollers, Version 2.1.24
    Copyright: (C) 2011-2015 Serge Vakulenko
\\.\COM17: Cannot open

No target found.
-- Running native process in default directory: "C:\Users\bschmalz\AppData\Local\Arduino15\packages\chipKIT\tools\pic32prog\v2.1.24\pic32prog.exe" -d "COM17" -b "115200" "C:/Users/bschmalz/MPLABXProjects/blink_WiFire_NoCopy.X/dist/default/production/blink_WiFire_NoCopy.X.production.hex" --
Programmer for Microchip PIC32 microcontrollers, Version 2.1.24
    Copyright: (C) 2011-2015 Serge Vakulenko
\\.\COM17: Cannot open

No target found.
-- Running native process in default directory: "C:\Users\bschmalz\AppData\Local\Arduino15\packages\chipKIT\tools\pic32prog\v2.1.24\pic32prog.exe" -d "COM17" -b "115200" "C:/Users/bschmalz/MPLABXProjects/blink_FBMini_NoCopy.X/dist/default/production/blink_FBMini_NoCopy.X.production.hex" --
Programmer for Microchip PIC32 microcontrollers, Version 2.1.24
    Copyright: (C) 2011-2015 Serge Vakulenko
\\.\COM17: Cannot open

No target found.

gholdys, 2017-02-15:
That's odd. I haven't changed anything in the regular MPLAB X "build" command. I'll try to reproduce this issue on my env.

EmbeddedMan, 2017-02-15:
It only happened to me after playing around with multiple sketches, all for
FB Mini (not that the particular board matters). It just seems to always
try to program after a build, and each time I do the build, it adds an
extra program attempt. (As you can see, I was up to 4 when I captured the
output.)

gholdys, 2017-02-15:
I think I fixed it. Please check: [Version 0.9.2](https://github.com/chipKIT32/chipKIT-importer/blob/master/dist/com-microchip-mplab-nbide-embedded-chipkit.nbm)

EmbeddedMan, 2017-02-16:
With version 0.9.2, I am no longer able to get this particular problem to happen.

However, I notice that it will do a build and then program every time I click the CK button. Is that the intended operation? I would have thought it would just be a program (no build).

gholdys, 2017-02-16:
That was the original behaviour (just program - no build) but later it was decided that it may be annoying for users to click "Build" and then "Program" every time they want to run code on their device so the "Program" button should also build. 
I can try to check modification dates on source files and the .hex file and run the build only if the source was modified after last compilation.

closed 2017-02-16 by gholdys