chipKIT32/chipKIT-importer #7: Mplab X doesn't know about certain Arduino variables/objects

EmbeddedMan, 2017-02-21:
The code helper thing (red underlines) in MPLAB X doesn't appear to be given the right paths for include files or something. When I import even a simple sketch like Blink, I see red underlines for the "ledPin" constant. If I import AnalogReadSerial, I get red underlines for the "Serial" object.

gholdys, 2017-02-21:
I think the problem appears when you try to open a source file right after importing when the IDE is still analysing the code. At least this is what I found out on my environment. The red underlines appeared only when I opened some file right after the wizard window disappeared. But if I waited a moment, the source files were parsed properly. Moreover, even when the red underline appeared, they disappeared after building. So the "includes" must have been set properly. It was just the source code parser that didn't make it.

EmbeddedMan, 2017-02-21:
That is not the experience I have had. I waited for quite a while (several
minutes), then built the code (it built without errors) and my red was
still there.

Not a big deal at all, and since it works for you, it probably will for
other folks,

gholdys, 2017-02-21:
Hold on... If the build failed, then it's a serious problem. Please give more details: What was the target board? What type of import did you try (with copying files or without)? What project did you try this on?

EmbeddedMan, 2017-02-21:
No, the build has no errors. It was the blink sketch for Fubarino Mini, both in Copy and NoCopy modes.

gholdys, 2017-02-21:
Thanks, I'll check this.

lstandage, 2017-02-21:
In the documentation of the UECIDE importer, it was indicated that certain IDE things had to be turned off due to this issue. Specifically, "Unresolved Identifier" and "Unresolved Include File".

Do we want to keep this documented this way for now?

EmbeddedMan, 2017-02-21:
Well, something isn't right with MPLAB X. Here's a screen snap of what I'm
seeing:

[image: Inline image 1]

If I hover over the lightbulbs, it says "unable to resolve identifier
ledPin", which is crazy - I'm defining it on line 2! Completely aside from
the importer, there's something messed up with X or with my settings here.

*Brian

EmbeddedMan, 2017-03-14:
I am unable to reproduce this with the latest (v0.9.10) importer plugin. I'm not sure why it's working now, but I tried several sketches (Blink, ReadAnalogSerial), for both Copy and NoCopy modes, and it appears work work just fine. So that makes me very happy. I'm going to close this and reopen it if it pops up again.

closed 2017-03-14 by EmbeddedMan