chipKIT32/chipKIT-importer #8: Serial Monitor: add ability to send bytes without clicking send

EmbeddedMan, 2017-02-22:
It would be a very nice feature (that even Arduino serial monitor doesn't have) to allow the user to type a single character and send that right off to the board, rather than wait for the user to also click Send.

There should be a way to choose which behavior mode you want to use.

gholdys, 2017-02-22:
Any idea how should we control this behaviour from the UI? Some checkbox, a toggle-button? I can't think of any good UI solution right now.

EmbeddedMan, 2017-02-22:
Well, I think it has to be mutually exclusive with the CR and LF buttons.

For example, here are the possible modes I can see:

Press send, with CR and LF
Press send, with CR
Press send, with LF
Press send
Don't press send

We already have the first four. The key is that you will never have the
case where you want to send each byte as you type (i.e. "don't press send")
but have the serial monitor add a CR or LF after each byte. So maybe there
could be another button, just like CR and LF, but that says "Use Send".
This button will be grayed out (not pressable) until both CR and LF are
unpressed. At that point, the "Use Send" button will become active.

This is just a quick guess at how to do it. Another way would be a checkbox
that says "Use Send", and the checkbox only becomes active when both CR and
LF are unpressed. Honestly, I think it would be easier to understand if CR
and LF were checkboxes as well.

gholdys, 2017-02-22:
OK and should there be an "echo" for the single-byte mode? Should the characters that the user types in appear in the messages view below? Should they disappear immediately from the input field or should they stay there?

EmbeddedMan, 2017-02-22:
Ahh, echo should be another another check box entirely! Good catch. The
echo check box should be available at all times to check or uncheck. If it
is checked, then everything you type in the 'send' text field will also be
printed to the output window. (either when you click Send, if "Use Send" is
checked, or as you type, if it is not. If the echo checkbox is unchecked,
then none of what you type gets put into the output text box.