chipKIT32/chipKIT-importer #10: Compiler requirement tied to test candidate version

lstandage, 2017-03-08:
The importer currently requires the following version of XC32:

1.43-TC5

But installing 1.43-RC1 gives an error.

It looks like Requirements.java has the version too long and needs to be shortened.

gholdys, 2017-03-08:
Fixed with version 0.9.11. Please verify.

EmbeddedMan, 2017-03-08:
Do we have a link for the new RC compiler?

Brian

guymc, 2017-03-08:
Uploading now, will appear here in about 5 minutes:

https://microchip.box.com/s/3rtrjbct27femocf6xyvfdsjxp633xrs

Guy

EmbeddedMan, 2017-03-08:
Interesting - I'm still using plugin version 0.9.9, and with the new
compiler, (1.43 RC1) I had no trouble importing or building a sketch.

Hmmm.

*Brian

lstandage, 2017-03-09:
Did you remove the previous compiler version?

EmbeddedMan, 2017-03-09:
Ahh, no. I did not. I just installed the new one, so my machine had both.

*Brian

lstandage, 2017-03-13:
The plugin now does work with the new compiler.

But a question.

From the code, it looks like the importer is checking the path of the compiler to see whether it is the correct version. Will this handle future versions of the compiler, say, 1.44?

How will it handle situations where the user has installed the compiler in a different directory than the default?

Not that we can cover all situations, but I wonder if we can get the version directly from the compiler itself.

Finally, how do we tell which compiler is valid? I think the problem I had earlier today was because MPLAB X had a v1.43-TC5 in its tools list, but it wasn't actually on the computer. Can we test the compiler before selecting it, or can MPLAB X tell us that there may be a problem with that tool?

lstandage, 2017-04-25:
Marking fixed unless proven otherwise.

closed 2017-04-25 by lstandage