chipKIT32/chipKIT-importer #12: Save files prior to rebuilding the Core

lstandage, 2017-03-14:
When invoking the "Rebuild Core Library" command, any files currently being edited should be saved prior to running the build.

gholdys, 2017-03-19:
I'm assuming here that by "any files" you mean core files. Is that right? If so then we should probably reconsider whether we want the core files to be editable at all. Especially in the no-copy import mode.

guymc, 2017-03-20:
As developers we may find the ability to rebuild the core quite handy. The average user won't need to, but the option is already out of the way (near the bottom of a long context menu.) I'm in favor of having the extra functionality accessible, and in helping the user avoid mysterious errors.

gholdys, 2017-03-23:
This functionality has been implemented in version 0.9.12. Please verify.

lstandage, 2017-06-05:
Fix confirmed. Closing.

closed 2017-06-05 by lstandage