chipKIT32/chipKIT-importer #14: Harmony Compatibility

peeern, 2017-04-07:
Sorry if it's not the proper place but I don't know where to ask questions about chipkit importer. Is there a way to make it work together with Harmony?

gholdys, 2017-04-07:
@guymc, could you comment on that?

EmbeddedMan, 2017-04-07:
My guess is that it will likely work OK with the harmony branch of
chipKIT-core. Keep in mind that we're not planning on continuing work on
the harmony branch of chipKIT-core as it wasn't getting us where we needed
to go. While we still have hope that someday we might be able to integrate
Harmony, in some form, with chipKIT-core, at this point in time we don't
have a technical path that gives us the features we want (high speed USB
and small size).

But as far as the plugin, I don't know that there's anything that special
about the harmony branch of chipKIT-core. You're welcome to try it out
yourself if you'd like. The plugin should just copy all of the files over
and create the project around them, which, for the harmony branch of
chipKIT-core, will be a _lot_ of files.

Note that this is just a guess - I haven't tried this.

*Brian

peeern, 2017-04-07:
Thank you for the quick replies. I apologize for the probable misunderstanding, but I wasn't aware of this branch you mentioned named chipkit-core-harmony. To be honest I was simply referring to MPLAB Harmony. My intention is to import my arduino based project into MPLAB and be able to make some changes using MPLAB Harmony. Do you think it's possible?

EmbeddedMan, 2017-04-07:
Peeern,

Well, the importer isn't set up to easily support that.

Now, it's all just C, right? So you could import an arduino project using
the importer, then add all of Harmony files yourself into that new project,
and then figure out how to integrate the two (main loop and such). That's
going to be a lot of work, but is technically possible.

*Brian