chipKIT32/chipKIT-importer #16: Fubarino Mini can't be debugged with importer

EmbeddedMan, 2017-04-23:
This is for importer v1.0.0

I'm trying to use my RealIce to program and debug a simple sketch. (Blink for example)

When I use the 'default' build configuration and my hardware programmer tool to debug, I get an error when programming that says "The requested operation cannot continue with the following configuration bit settings(s): Watchdog Timer Enable = On" and then it gives you the option to override this bit for just this program cycle.

This should not come up - we should be sure that the bootloader-stub we're using when hardware programming has the proper configuration bits set.

Also, I'm unable to get my sketch to run or debug when I use the hardware debugger in either default configuration. We aren't expecting that to work, are we?

guymc, 2017-04-23:
At this point, we expect the importer to be fully functional with the Fubarino Mini, creating and downloading production builds, and debugging with either the simulator or a hardware debugger. If that's not the case, then we have some more work to do.

Incidentally, I recall that bootloaders for both 40MHz and 48MHz were created. But I can't say for sure at the moment, that the production boards were intended to run at 48Mhz. That would have required us to use 50MHz-rated devices, which are less common. Larry should be able to get this answer on Monday.

EmbeddedMan, 2017-04-23:
Ahh. I see. I vaguely remember us deciding to use the 50MHz parts for production so that we could run at 48Mhz. But I may be remembering that wrong.

EmbeddedMan, 2017-04-24:
I checked with 2 Fubarino Minis that I still have in factory fresh packaging from Microchip - and they are both tho -50 (50MHz) variants, with the 48MHz bootloader onboard. So that is what we should be using for the importer.

lstandage, 2017-04-24:
Was there a "debug" configuration? That's the one that should be used with
the RI/ICD3/PK3.

The default configuration is intended only for the "cK" button, with
pic32prog.

EmbeddedMan, 2017-04-24:
Right - yes, there is a debug configuration, and when I use it to debug with the RealIce, the project does work properly.

OK, I see now. You do a normal Make and Program device when in Debug configuration to get the bootloader back on, then Default configuration for building and programming with pic32Prog.

That all seems to work fine. So the main point of this issue is not valid - it was a mistake on my part.

closed 2017-04-24 by EmbeddedMan