chipKIT32/chipKIT-importer #19: Created project uses absolute paths instead of relative ones

lstandage, 2017-04-25:
When the importer creates the project, it puts all of the files in absolute locations. This leads to not being able to, say, package the project and try it out on either a different platform or in a different directory/drive.

Relative paths should be used in order to make the project platform-independent.

gholdys, 2017-04-26:
This will be possible only for the "copy-all" import mode, but it is definitely worth implementing.

gholdys, 2017-05-04:
Fix implemented in version 1.0.1. Please verify if this issue has been successfully resolved.

EmbeddedMan, 2017-05-18:
It looks like .ld files still have absolute paths, not relative ones.

I imported a Fubarino Mini sketch at my home machine, it compiled and worked great. Then I zipped up the project folder, moved it to a different PC (with a different username), unzipped it and tried to build it and it failed because the chipKIT-application-32MX250F128-debug.ld has an absolute path and so isn't found.

EmbeddedMan, 2017-05-18:
And the files as listed in the Project explorer are correct. It's just the linker command line that has the path wrong:

make -f nbproject/Makefile-debug.mk SUBPROJECTS= .build-conf
make[1]: Entering directory 'C:/Users/bschmalz/MPLABXProjects/hellodotstar.X'
make  -f nbproject/Makefile-debug.mk dist/debug/production/hellodotstar.X.production.hex
make[2]: Entering directory 'C:/Users/bschmalz/MPLABXProjects/hellodotstar.X'
"C:\Program Files (x86)\Microchip\xc32\v1.43\bin\xc32-g++.exe" -g -x c++ -c -mprocessor=32MX250F128D -IchipKIT-core -IchipKIT-libraries/Adafruit_DotStar -IchipKIT-libraries/SPI -frtti -fno-exceptions -fno-check-new -fenforce-eh-specs -DF_CPU=48000000L -DARDUINO=10613 -D_BOARD_FUBARINO_MINI_ -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino -DXPRJ_default=default -D__CTYPE_NEWLIB -MMD -MF build/debug/production/chipKIT-libraries/SPI/SPI.o.d -o build/debug/production/chipKIT-libraries/SPI/SPI.o chipKIT-libraries/SPI/SPI.cpp -DXPRJ_debug=debug -DARDUINO_ARCH_PIC32 -mno-smart-io -ffunction-sections -fdata-sections -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -G1024 -D_USE_USB_FOR_SERIAL_ -mnewlib-libc -std=gnu++11 
"C:\Program Files (x86)\Microchip\xc32\v1.43\bin\xc32-g++.exe" -g -x c++ -c -mprocessor=32MX250F128D -IchipKIT-core -IchipKIT-libraries/Adafruit_DotStar -IchipKIT-libraries/SPI -frtti -fno-exceptions -fno-check-new -fenforce-eh-specs -DF_CPU=48000000L -DARDUINO=10613 -D_BOARD_FUBARINO_MINI_ -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino -DXPRJ_default=default -D__CTYPE_NEWLIB -MMD -MF build/debug/production/source/hellodotstar.o.d -o build/debug/production/source/hellodotstar.o source/hellodotstar.cpp -DXPRJ_debug=debug -DARDUINO_ARCH_PIC32 -mno-smart-io -ffunction-sections -fdata-sections -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -G1024 -D_USE_USB_FOR_SERIAL_ -mnewlib-libc -std=gnu++11 
"C:\Program Files (x86)\Microchip\xc32\v1.43\bin\xc32-g++.exe" -g -x c++ -c -mprocessor=32MX250F128D -IchipKIT-core -IchipKIT-libraries/Adafruit_DotStar -IchipKIT-libraries/SPI -frtti -fno-exceptions -fno-check-new -fenforce-eh-specs -DF_CPU=48000000L -DARDUINO=10613 -D_BOARD_FUBARINO_MINI_ -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino -DXPRJ_default=default -D__CTYPE_NEWLIB -MMD -MF build/debug/production/chipKIT-libraries/Adafruit_DotStar/Adafruit_DotStar.o.d -o build/debug/production/chipKIT-libraries/Adafruit_DotStar/Adafruit_DotStar.o chipKIT-libraries/Adafruit_DotStar/Adafruit_DotStar.cpp -DXPRJ_debug=debug -DARDUINO_ARCH_PIC32 -mno-smart-io -ffunction-sections -fdata-sections -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -G1024 -D_USE_USB_FOR_SERIAL_ -mnewlib-libc -std=gnu++11 
"C:\Program Files (x86)\Microchip\xc32\v1.43\bin\xc32-g++.exe"   -mprocessor=32MX250F128D -mdebugger -mno-peripheral-libs -nostartfiles -mnewlib-libc -T"C:\Users\Brian Schmalz\MPLABXProjects\hellodotstar.X\chipKIT-core\chipKIT-application-32MX250F128-debug.ld" -T"C:\Users\Brian Schmalz\MPLABXProjects\hellodotstar.X\chipKIT-core\chipKIT-application-COMMON.ld" -o dist/debug/production/hellodotstar.X.production.elf build/debug/production/chipKIT-libraries/Adafruit_DotStar/Adafruit_DotStar.o build/debug/production/chipKIT-libraries/SPI/SPI.o build/debug/production/source/hellodotstar.o          -DXPRJ_debug=debug    -Wl,--defsym=__MPLAB_BUILD=1,--gc-sections,--no-code-in-dinit,--no-dinit-in-serial-mem,-Map="dist/debug/production/hellodotstar.X.production.map",--memorysummary,dist/debug/production/memoryfile.xml,-LchipKIT-core,-lCore
xc32-cpp.exe: error: C:\Users\Brian Schmalz\MPLABXProjects\hellodotstar.X\chipKIT-core\chipKIT-application-32MX250F128-debug.ld: No such file or directory
xc32-cpp.exe: warning: '-x c' after last input file has no effect
xc32-cpp.exe: fatal error: no input files
compilation terminated.
xc32-cpp.exe: error: C:\Users\Brian Schmalz\MPLABXProjects\hellodotstar.X\chipKIT-core\chipKIT-application-COMMON.ld: No such file or directory
xc32-cpp.exe: warning: '-x c' after last input file has no effect
xc32-cpp.exe: fatal error: no input files
compilation terminated.
c:\program files (x86)\microchip\xc32\v1.43\bin\bin\gcc\pic32mx\4.8.3\..\..\..\..\bin/xc32-ld.exe: cannot open linker script file chipKIT-application-32MX250F128-debug.ld.00: No such file or directory
collect2.exe: error: ld returned 255 exit status
make[2]: *** [dist/debug/production/hellodotstar.X.production.hex] Error 255
make[1]: *** [.build-conf] Error 2
make: *** [.build-impl] Error 2
nbproject/Makefile-debug.mk:163: recipe for target 'dist/debug/production/hellodotstar.X.production.hex' failed
make[2]: Leaving directory 'C:/Users/bschmalz/MPLABXProjects/hellodotstar.X'
nbproject/Makefile-debug.mk:90: recipe for target '.build-conf' failed
make[1]: Leaving directory 'C:/Users/bschmalz/MPLABXProjects/hellodotstar.X'
nbproject/Makefile-impl.mk:39: recipe for target '.build-impl' failed

BUILD FAILED (exit value 2, total time: 715ms)

(The right path is "C:\Users\bschmalz\MPLABXProjects\hellodotstar.X\chipKIT-core\chipKIT-application32MX250F128-debug.ld")

gholdys, 2017-05-19:
Looks like the paths to .ld scripts aren't all that important. Only the file names need to be in the "-T" options. The directories that contain those scripts are already in the "-L" option.

EmbeddedMan, 2017-05-19:
OK, well, whatever needs to be changed so that the compile will succeed!

gholdys, 2017-05-19:
I've created a branch for version 1.0.5. We can test all the new features and fixes on the branch before releasing the new version. The new plugin package is [here](https://github.com/chipKIT32/chipKIT-importer/blob/v1.0.5/dist/com-microchip-mplab-nbide-embedded-chipkit.nbm)

EmbeddedMan, 2017-05-19:
That appears to have fixed the problem. With the new v1.0.5, I can zip up a project directory, move it to another machine (with a different user file structure) and it builds just fine. 

Thanks!!!

closed 2017-05-19 by EmbeddedMan