chipKIT32/chipKIT-importer #21: Installation Instructions

shibshib, 2017-05-04:
It's not clear whether this should be installed as a third party plugin or from the plugin manager. If installing thru the plugin manager, JNA dependency errors prevent the installation. 

[image]

shibshib, 2017-05-04:
Nevermind, just used the .nbm file to import manually. Thanks.

closed 2017-05-04 by shibshib

guymc, 2017-05-04:
Please note exactly which version of MPLAB X you are using.

We installed a work-around for the dependency issue (or so we thought.) A permanent fix is waiting for the next update of X.

Thanks!

Closed #21<https://github.com/chipKIT32/chipKIT-importer/issues/21>.

shibshib, 2017-05-04:
For anyone else experiencing this issue: I was using MPLAB X v. 3.60. The workaround i did was download this repository and manually install the plugin from the plugin manager using the .nbm file included in this repository.