chipKIT32/chipKIT-importer #22: Cannot find Chipkit importer on available plugins.

annerajb, 2017-05-07:
I downloaded MP LAB X IDE  v3.51 and there is no import -> import chipkit menu item.

Here is a picture of the plugins repository

[image]

In the Available or installed tabs I cannot find kit typying multiple variations in the search like, arduino, sketch, import, chipkit

Are the instructions missing a step required for this plugin to be installed? Like adding the plugin repository where it resides?

Also there are no instructions on how to compile from source Would be nice adding 2-4 lines explaining how to build the plugin from source.

EmbeddedMan, 2017-05-07:
I _think_ you have to use 3.60 to get it to appear in the default plugin list.

If you want to download it manually, click this link : https://github.com/chipKIT32/chipKIT-importer/raw/master/dist/com-microchip-mplab-nbide-embedded-chipkit.nbm

Then you can install that .nbm file from the Tools->Plugins->Downloaded->Add Plugins ... button.

gholdys, 2017-05-07:
@annerajb, I've added some instruction about building from source to the README.MD file. Let me know if it's sufficient.

annerajb, 2017-05-07:
Thanks. I had installed 3.60 but my shortcut was still pointing to the old version.
I can now select the plugin but it gives a error which seems to be related to the IDE since I cannot find a JNA plugin on the list. 
[image]

gholdys, 2017-05-11:
The newest version of the plugin - available on GitHub in the "dist" directory and soon in the plugin repository - should install correctly on MPLAB X IDE v3.60 and newer

annerajb, 2017-05-11:
Thanks I was able to install it fine now. 

Now it's giving me a error this is the same error i got with the built from source version a few days ago
[image]
[image]

lstandage, 2017-05-11:
The chipKIT Cmod has a bootloader that isn't supported at the moment.
Internally, the plugin has a "database" matching the selected board to a
bootloader file. This one doesn't have it, so it generates the error.

I'll see if there's a way to work around it.

annerajb, 2017-05-28:
Any ideas?

lstandage, 2017-06-14:
It looks like the current version of the plugin (v1.0.4) should provide the correct bootloader. But we are also in the process of getting v1.0.5 ready to go. Can you confirm whether it is still a problem?

Also, because of a problem I worked on yesterday, you might need to verify the version of XC32 that you're using (v1.43 or later).

If you would, turn on logging in MPLAB X, (under Tools->Other, Embedded->Diagnostics tab, Logging Level drop-down box). Then try the plugin again, and view the log through the View->IDE Log menu. With that, we'll be able to tell what exactly is going on.

annerajb, 2017-06-22:
I will verify see if I can debug tonight. XC32 should be really recent (withing a month)

Thanks