chipKIT32/chipKIT-importer #24: Need to default to -O2

EmbeddedMan, 2017-06-06:
Arduino, by default (I believe) uses -O2 on it's command line to the compiler.

If we are to mimic the performance of the Arduino IDE built sketches as closely as possible, we'll need to do the same thing.

Right now we do not have any optimization level set on our command line. It should be -O2 like Arduino's.

gholdys, 2017-06-06:
`"Compiler option (Optimization level > 1) ignored because the free XC32 C++ compiler does not support this feature."`

This is what you get when you try to set the optimization level to something larger than 1 on the free compiler. I can change the default optimization level to 1 but I don't think that it makes sense to set it to 2 (although maybe it is possible to automagically recognize whether the user has a full version of the XC32 or just the freeware one)

EmbeddedMan, 2017-06-06:
I thought that we used the installed chipkit compiler for the plugin. If
not, we should. It has full optimizations.

Brian

guymc, 2017-06-06:
Actually, we decided to use the standard XC32 compiler, with the open-source libraries option selected. In the default import mode, all source files and libraries are copied into a stand-alone MPLAB X project. Being stand-alone, the project should not have any dependencies on chipKIT-core, including the compiler.

GH has the right idea. Imported projects should default to -O1 optimization, in order to avoid the annoying message. On the other hand, this automatically short-changes someone who has paid for a compiler license that provides full optimizations. I confirmed with Jason K. this morning, that level 1 optimizations will be implemented if you pass -O2 on a free compiler. And, you get the annoying Warning message.

All things considered, I think we should stick with GH's idea, and set the importer to use -O1. If someone pays for a compiler license, they can increase the optimization level manually, in their Project settings.

I wonder what tangible impact the difference between -O1 and -O2 will be? My demo machine uses the free compiler, so I'll find out soon enough how -O1 impacts the PICxel library and NeoPixel demos that Brian has been working on.

Guy

EmbeddedMan, 2017-06-06:
Guy, the differences in timing between O1 and O2 are not small, but your
Neopixels will still work. This is too bad that Arduino IDE gives faster
and smaller code than X without buying a compiler

Brian

guymc, 2017-06-06:
Thanks; I agree with your point. It doesn't seem right that Arduino IDE gives faster
and smaller code than X without buying a compiler.

One possible solution: In MPLAB X:Tools:Options:Embedded:Build Tools, add the chipKIT compiler as a "Custom Compiler". This would not be convenient, because you'd have to go hunting for the chipKIT compiler. Also, we couldn't recommend this too broadly, because it discounts a policy about compiler licensing.

Any other ideas?

commit 26faefbc (2017-06-10): Fixes #24: Sets default optimization level to 1
closed 2017-06-10 by gholdys