chipKIT32/chipKIT-importer #26: chipkit plugin java exception

parmiani, 2017-06-22:
Hi everybody,
I have a problem with chipkit plugin on my Mplab x 3.61 (with xc32-1.43 compiler). The plugin is installed correctly, but when i select "import Chipkit project" i receive a  java.lang.NullPointerException and i cannot proceed . I attach the log file of Mplab x
[messages.txt](https://github.com/chipKIT32/chipKIT-importer/files/1096381/messages.txt).

Someone can help me?

Thanks a lot

Alessandro

EmbeddedMan, 2017-06-22:
What version of the plugin do you have, and what OS are you running?

parmiani, 2017-06-23:
I have downloaded the latest version on git hub. The operating system is windows 7

parmiani, 2017-06-23:
The plugin version is 1.0.5

gholdys, 2017-06-23:
It seems that the plugin cannot find the chipKIT platform files. Are you sure you have installed the chipKIT package for Arduino?

gholdys, 2017-06-24:
Of course, there's definitely an error in the plugin which should check the existence of the Arduino/chipKIT platform files and provide a meaningful error message to the user. We will implement this functionality in the next release (1.0.6)

parmiani, 2017-06-24:
I tried to reinstall the Arduino package and after to reinstall the plugin and now seems that all works correctly. Probably plugin  doesn't see the Arduino files.

Thank you very much for your help.

Alessandro

lstandage, 2017-08-22:
Closing issue.

closed 2017-08-22 by lstandage