chipKIT32/chipKIT-importer #30: Can't debug imported projects on custom MAX32-like board

yoalieh, 2017-11-13:
Hi, I'm having a long weekend because of this problem, and I'm not pretty sure I can explain it very well.

I have two machines, one with
MplabX 4.01, xc-32 1.44, arduino 1.8.5 and CK_core 1.4.3, importer 1.0.6
The other one with:
MplabX 3.54, xc-32 1.43, arduino 1.8.1 and CK_Core 1.4.3, importer 1.0.6

I have a complex project including CK Libraries DEIPCK, EEPROM, DSPI and IM8720PHY

The board I'm using is a custom board almost identical to MAX32, with no serial output.

I compiled and uploaded the imported project to the board, and it works, but when I try to debug, it won't. I set the right configuration bits using MPLABX, and the imported project couldn't run in debug mode, it gives back:

> The  target device is not ready for debugging. Please check your configuration bit settings and program the device before proceeding. The most common causes for this failure are oscillator and/or PGC/PGD settings.

I confirm the configuration bits are ok, and the debugging capabilities using an ICD3 and a clean project created in MPLABX with the same configuration bits.

I then tested a clean sketch, empty loop and setup, and imported the project in MPLABX, I compiled and uploaded, and it works, but I can't debug. I add the configuration bits, and still can't debug.

Does this happens to others? Are there any suggestions on what to do or test next? 

I'm currently downgrading one of the machines's software to the minimum recommended by the importer. I also can test directly to a max32, though I need to solder debug header.

yoalieh, 2017-11-13:
Tested with the MAX32 and it worked right, the trouble seems to be the difference between boards, as custom board uses the PGED1 and PGEC1, both of which are A0 and A1 in the MAX32, thus they are being used as analog.

yoalieh, 2017-11-13:
Just in case some one needs to know, the trouble was in fact the PGED and PGEC terminals, I originally was trying to set them with #pragma, but the .ld script fetched from CK core by the import tool takes precedence when compiling, and I had to set the bits manually in chipKIT-application-32MX795F512-nobootloader.ld in section

.config_BFC02FFC : {
    LONG(0x7FFFFFF6)
  } > config0

closed 2017-11-13 by yoalieh

gholdys, 2017-11-13:
Hello

Sorry for the late reply. This is an interesting case that you've run into. I'm glad you solved it and provided the solution as it would most likely take us more time to figure it out.

Cheers