chipKIT32/chipKIT-importer #32: Import issue of chipKIT-importer V1.0.6 with MPLAB X IDE

hxfeng1978, 2018-07-04:
I have installed MPLAB X IDE V4.15, chipKIT Import plugin V1.0.6 and chipKIT-core V1.4.1 / V2.0.5. But when I  launched chipKIT Import and import a Sketch, the import operation is always faied. 

The import fail information as  shown below:
java.lang.NullPointerException
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.importChipKitProjectFiles(ImportWorker.java:264)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.createProjectFromChipKit(ImportWorker.java:189)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.invokeImporterTasks(ImportWorker.java:144)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.doInBackground(ImportWorker.java:94)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.doInBackground(ImportWorker.java:72)
	at javax.swing.SwingWorker$1.call(SwingWorker.java:295)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at javax.swing.SwingWorker.run(SwingWorker.java:334)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)

I heard that other people have used the chipKIT Import plugin V1.0.5 to import successfully. I also want to try chipKIT Import plugin V1.0.5. So, could you please provide chipKIT Import plugin V1.0.5 install package? Thanks!

gholdys, 2018-07-04:
I'm on it. Thanks for the bug report.

gholdys, 2018-07-15:
hxfeng1978, I'm sorry it took so long but there seems to be a more complicated problem here. You can get version 1.0.5 of the plugin here:
https://github.com/chipKIT32/chipKIT-importer/blob/aa3ce30b3a7de263571e8d9f8b6a6d88eca1ad07/dist/com-microchip-mplab-nbide-embedded-chipkit.nbm
However, frankly, I'm not convinced if this will actually solve your problem. Please try and let me know.