chipKIT32/chipKIT-importer #33: Importing with chipKIT-importer v1.0.6 - NullPointerException

cpenny42, 2018-08-20:
I am trying to import the basic `Blink.ino` example, but the plugin always crashes. This happens with both MPLAB X IDE V4.15 and MPLAB X IDE v5.0.5, Importer v1.0.6 and chipKIT-core v2.0.5.

There are two different messages depending on whether "Copy all chipKIT Dependencies" is checked.

With "Copy all chipKIT Dependencies" ON:
```
java.lang.NullPointerException
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.addFileToFolder(ImportWorker.java:434)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.lambda$importChipKitProjectFiles$7(ImportWorker.java:318)
	at java.util.stream.ForEachOps$ForEachOp$OfRef.accept(ForEachOps.java:184)
	at java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:175)
	at java.util.Iterator.forEachRemaining(Iterator.java:116)
	at java.util.Spliterators$IteratorSpliterator.forEachRemaining(Spliterators.java:1801)
	at java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:481)
	at java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:471)
	at java.util.stream.ForEachOps$ForEachOp.evaluateSequential(ForEachOps.java:151)
	at java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(ForEachOps.java:174)
	at java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.util.stream.ReferencePipeline.forEach(ReferencePipeline.java:418)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.importChipKitProjectFiles(ImportWorker.java:317)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.createProjectFromChipKit(ImportWorker.java:189)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.invokeImporterTasks(ImportWorker.java:144)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.doInBackground(ImportWorker.java:94)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.doInBackground(ImportWorker.java:72)
	at javax.swing.SwingWorker$1.call(SwingWorker.java:295)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at javax.swing.SwingWorker.run(SwingWorker.java:334)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
```

And with "Copy all chipKIT Dependencies" OFF:
```
java.lang.NullPointerException
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.importChipKitProjectFiles(ImportWorker.java:337)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.createProjectFromChipKit(ImportWorker.java:189)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.invokeImporterTasks(ImportWorker.java:144)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.doInBackground(ImportWorker.java:94)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.doInBackground(ImportWorker.java:72)
	at javax.swing.SwingWorker$1.call(SwingWorker.java:295)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at javax.swing.SwingWorker.run(SwingWorker.java:334)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
```

I have yet to see a working example of the Importer for any kind of sketch — can anyone point me in the right direction? Even just having an example of an MPLAB project made from a successfully imported sketch would help.

I tried importing a project with version 1.0.5 [posted on this issue](https://github.com/chipKIT32/chipKIT-importer/issues/32), and had a similar crash (though the line numbers are different):
```
java.lang.NullPointerException
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.addFileToFolder(ImportWorker.java:427)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.lambda$importChipKitProjectFiles$7(ImportWorker.java:313)
	at java.util.stream.ForEachOps$ForEachOp$OfRef.accept(ForEachOps.java:184)
	at java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:175)
	at java.util.Iterator.forEachRemaining(Iterator.java:116)
	at java.util.Spliterators$IteratorSpliterator.forEachRemaining(Spliterators.java:1801)
	at java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:481)
	at java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:471)
	at java.util.stream.ForEachOps$ForEachOp.evaluateSequential(ForEachOps.java:151)
	at java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(ForEachOps.java:174)
	at java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.util.stream.ReferencePipeline.forEach(ReferencePipeline.java:418)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.importChipKitProjectFiles(ImportWorker.java:312)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.createProjectFromChipKit(ImportWorker.java:184)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.invokeImporterTasks(ImportWorker.java:141)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.doInBackground(ImportWorker.java:92)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.doInBackground(ImportWorker.java:72)
	at javax.swing.SwingWorker$1.call(SwingWorker.java:295)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at javax.swing.SwingWorker.run(SwingWorker.java:334)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
```

cpenny42, 2018-08-21:
I pulled the repo & got started with the MPLAB SDK in Netbeans and was able to get some more clues. I'm unfamiliar with Java development/Netbeans/MPLAB projects, so there are probably some obvious things I am overlooking. 

The problem seemed to stem from [this line](https://github.com/chipKIT32/chipKIT-importer/blob/master/src/com/microchip/mplab/nbide/embedded/chipkit/wizard/ImportWorker.java#L434) in ImportWorker.java: 
`Path projectRootPath = Paths.get( folder.getConfigurationDescriptor().getBaseDir() );`
As it turns out, the `folder` was null.

The `folder` parameter was passed in to `addFileToFolder` from `importChipKitProjectFiles` above [here at line 315 of ImportWorker.java](https://github.com/chipKIT32/chipKIT-importer/blob/master/src/com/microchip/mplab/nbide/embedded/chipkit/wizard/ImportWorker.java#L315).

This is the problematic line:
`Folder sourceFolder = newProjectDescriptor.getLogicalFolders().findFolderByName(MakeConfigurationBook.SOURCE_FILES_FOLDER);`

`findFolderByName` was returning `null` when looking for `MakeConfigurationBook.SOURCE_FILES_FOLDER`. So `sourceFolder` was null when it was passed into `addFileToFolder`, causing the specific assertion I saw.

With some further prodding I discovered that `newProjectDescriptor.getLogicalFolders()` would return a `Folder` object with only 2 subfolders: `chipKIT-libraries` and `chipKIT-core`. This is a problem because the actual string returned from `MakeConfigurationBook.SOURCE_FILES_FOLDER` is simply the string `"SourceFiles"`, which is not one of the subfolders returned by `newProjectDescriptor.getLogicalFolders()`. I'm assuming `findFolderByName` just searches for a sub-folder with that name; if so this error would make sense, since it's searching for the folder `"SourceFiles"` when that folder doesn't exist in the `MakeConfigurationBook`. Not completely sure though.

I'm not exactly sure what the folders are supposed to represent inside a `MakeConfigurationBook`, but I'm assuming it's just the normal mplab project folders we're trying to create during the import.

I was able to get the plugin to complete without crashing, but the generated project was completely empty. I'm guessing there is some issue with finding the correct source folder to copy over.

gholdys, 2018-08-21:
Hi Chris

You are generally right - there is an issue with the source folder. It seems that in MPLAB X IDE v5.0.x the source directory is not created by default like it was in the earlier versions. I'll add some logic that adds this directory if it does not exist.

Regards,
Grzegorz

gholdys, 2018-08-21:
@cpenny42, I've pushed version 1.0.7 to the repo. You will find the new .nbm file here: https://github.com/chipKIT32/chipKIT-importer/blob/master/dist/com-microchip-mplab-nbide-embedded-chipkit.nbm
You will still get an error message at the end of the import procedure but the project should be created inside MPLAB X IDE and it should compile (at least this is what I got)

cpenny42, 2018-08-22:
Thanks! I'll check it out - this is a huge help, I appreciate it!

closed 2018-08-22 by cpenny42

cpenny42, 2018-08-22:
Everything worked perfectly with the update!