chipKIT32/chipKIT-importer #35: mreserve options in my DEBUG builds failed at the linking stage

drewtenn, 2019-02-02:
I imported an Arduino project for my Chipkit MAX32 board and running a production build worked fine. My DEBUG builds failed with "collect2: error: ld terminated with signal 6 [Abort trap: 6]"

In digging into the command that was causing it to fail, I narrowed it down to the mreserve options that were created in Makefile-default.mk.

-mreserve=data@0x0:0x1FC -mreserve=boot@0x1FC02000:0x1FC02FEF -mreserve=boot@0x1FC02000:0x1FC024FF

Removing these from the makefile allowed the build to succeed and I was then able to debug code running on the board.

1. Where do these values come from?
2. What/if are these necessary?

gholdys, 2019-02-03:
@drewtenn, those options seem to come from MPLAB X IDE itself. They don't appear in the chipKIT files (for Arduino) so cannot be imported.
@EmbeddedMan, do you have an idea what they are?

EmbeddedMan, 2019-02-03:
Unfortunately I do not. I've not seen them before. @jasonkajita  do you have any idea?