chipKIT32/chipKIT-importer #37: Java Exception on Windows7, MPLABX 5.25, Arduino 1.8.10

tsayman, 2019-12-02:
I have tried several times with the same results. 
Updated everything to the latest, tried again, etc...

Log file attached and clipboard paste of the error message below.

[messages.zip](https://github.com/chipKIT32/chipKIT-importer/files/3912131/messages.zip)

java.lang.RuntimeException: java.lang.NullPointerException
	at com.microchip.mplab.nbide.embedded.chipkit.importer.ArduinoBuilderRunner.preprocess(ArduinoBuilderRunner.java:109)
	at com.microchip.mplab.nbide.embedded.chipkit.importer.ArduinoBuilderRunner.preprocess(ArduinoBuilderRunner.java:84)
	at com.microchip.mplab.nbide.embedded.chipkit.importer.ChipKitProjectImporter.preprocessSourceProject(ChipKitProjectImporter.java:307)
	at com.microchip.mplab.nbide.embedded.chipkit.importer.ChipKitProjectImporter.execute(ChipKitProjectImporter.java:174)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.importChipKitProjectFiles(ImportWorker.java:275)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.createProjectFromChipKit(ImportWorker.java:189)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.invokeImporterTasks(ImportWorker.java:144)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.doInBackground(ImportWorker.java:94)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.doInBackground(ImportWorker.java:72)
	at javax.swing.SwingWorker$1.call(SwingWorker.java:295)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at javax.swing.SwingWorker.run(SwingWorker.java:334)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at com.microchip.mplab.nbide.embedded.chipkit.importer.ArduinoBuilderRunner.lambda$findMainLibraryPaths$0(ArduinoBuilderRunner.java:191)
	at java.lang.Iterable.forEach(Iterable.java:75)
	at java.util.Collections$UnmodifiableCollection.forEach(Collections.java:1080)
	at com.microchip.mplab.nbide.embedded.chipkit.importer.ArduinoBuilderRunner.findMainLibraryPaths(ArduinoBuilderRunner.java:186)
	at com.microchip.mplab.nbide.embedded.chipkit.importer.ArduinoBuilderRunner.preprocess(ArduinoBuilderRunner.java:99)
	... 14 more

gholdys, 2019-12-07:
Looking into it

g8kig, 2020-08-05:
Happens to me MPLABX v5.40. Arduino IDE 1.8.12, Windows 10

java.lang.RuntimeException: java.lang.NullPointerException
	at com.microchip.mplab.nbide.embedded.chipkit.importer.ArduinoBuilderRunner.preprocess(ArduinoBuilderRunner.java:109)
	at com.microchip.mplab.nbide.embedded.chipkit.importer.ArduinoBuilderRunner.preprocess(ArduinoBuilderRunner.java:84)
	at com.microchip.mplab.nbide.embedded.chipkit.importer.ChipKitProjectImporter.preprocessSourceProject(ChipKitProjectImporter.java:307)
	at com.microchip.mplab.nbide.embedded.chipkit.importer.ChipKitProjectImporter.execute(ChipKitProjectImporter.java:174)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.importChipKitProjectFiles(ImportWorker.java:275)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.createProjectFromChipKit(ImportWorker.java:189)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.invokeImporterTasks(ImportWorker.java:144)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.doInBackground(ImportWorker.java:94)
	at com.microchip.mplab.nbide.embedded.chipkit.wizard.ImportWorker.doInBackground(ImportWorker.java:72)
	at javax.swing.SwingWorker$1.call(SwingWorker.java:295)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at javax.swing.SwingWorker.run(SwingWorker.java:334)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at com.microchip.mplab.nbide.embedded.chipkit.importer.ArduinoBuilderRunner.lambda$findMainLibraryPaths$0(ArduinoBuilderRunner.java:191)
	at java.lang.Iterable.forEach(Iterable.java:75)
	at java.util.Collections$UnmodifiableCollection.forEach(Collections.java:1080)
	at com.microchip.mplab.nbide.embedded.chipkit.importer.ArduinoBuilderRunner.findMainLibraryPaths(ArduinoBuilderRunner.java:186)
	at com.microchip.mplab.nbide.embedded.chipkit.importer.ArduinoBuilderRunner.preprocess(ArduinoBuilderRunner.java:99)
	... 14 more